TinkerGnome/Ultimaker2Marlin #84: Total Power Budget ? TinkerGnome

Printo1987, 2017-12-08:
Hello today i instaled TinkerGnome and there is one option on Screen Ulti.   Power Budget , Extruder Watt ,Bed Watt and Total . with one watts i must write there?? I've extruder 30W  bed 160  and Supply Power 12.4Amper.

Thank you for help

gr5, 2017-12-08:
If you have a standard UM2 with the bed that came with the machine and the nozzle is<= 50 watts then you don't have to restrict any power.  So make sure the power is > the sum of bed and nozzle put together.

This power budget is for people who have for example 2 35W heaters on the head.  Or a um2go with a heater.  It tells the software to restrict the total power so the power brick doesn't trip an alarm and shut off.

For those mods, if you don't set a power budget low enough and the system goes for 100% on all heaters (beds and nozzles) will occasionally shut down the power.

Printo1987, 2017-12-08:
ok thank you for information.  I've  reprap ultimaker and parts its nor original   . I have one heater 30W   bed 160W  and power supply 295W   and with one parametrs i must put this option??  This is important for me Mr.

gr5, 2017-12-08:
30W + 160W = 190W.  That leaves you another 105W for everything else (steppers, fans, computer) which should be plenty.  So I don't think you need to worry about this feature.

Printo1987, 2017-12-08:
you mean:

Total Power Budget = 190W  or 105W ?
Bed=160W
Extruder =30W

Thank you For Help Mr.

gr5, 2017-12-08:
If power budget is greater than bed+extruder power then this code will not do anyting.  In other words it will not limit the power.  Which is probably what you want.  So just ignore all 3 values as long as budget is greater than bed + extruder.

Only if later you realize your power supply turns itself off, then you can fill in these values properly.  But I doubt this will happen.  So go ahead and make bed=1 watt, extruder=1 watt and power budget = 200 watts.  Or just leave it as is.  As long as budget is greater than bed+extruder.

Printo1987, 2017-12-08:
ok great... But Software know everything about my Watt Extruder and Watt Bed and Watt other parts .   Maybe software give to my extruder propably 50W ? or this is not possible?:)

Thanks 👍

gr5, 2017-12-09:
The software can turn the bed on and off 20X per second.  If the software decides to limit the power, it can do that by turning the bed on for only a short time each second.

This whole "power budget" is a feature that I don't think you need so I don't think you will understand it.  It is for other people with weak power supplies.

Printo1987, 2017-12-10:
ok m8.  Thank you :)  and tell me tinder software will be better at original software?? becuase last week i buy  quiet SUNON fun  10m3/h   and this fans not working  if i put 100%  is ok but if i put 20% fan or 50% fan pwm not working... and my friend check on the prusa i3 my fans coling and on the Prusa normal working 10% or 50% etc . you know whats going on ?:(

gr5, 2017-12-11:
We are getting way off topic - this should be on the forums.  But because computers can be smaller than a grain of rice and cost less than one penny, some fans now have computers in them.  These computers try to make the fan spin at the same speed even if the power is being turned on and off 20X per second (that's what Ultimaker does - turns fan on and off 20X per second to control the speed).  The manufacturer thinks this is a good feature but it is a problem when attaching to an ultimaker.  You want to buy a fan without a computer in it if you want to be able to control the speed.  If 100% is on, 0% is off and 50% is off then the tiny computer probably doesn't have time to boot up.  If 50% is on full, then the computer is able to pull the power it needs.  These fans with computers do unpredictable things.

Printo1987, 2017-12-11:
thanks you for help Mr :)

one question :) with one fans you use to Ultimaker :)

diettelo, 2017-12-27:
i have  an ultimaker 2 upgraded to 2+
with original heat bed and 50 w heater nozzle upgraded. so 50w and 160w total 210w?
Also i got a new stronger Power supply wih 15A and 24 Volt

gr5, 2017-12-27:
That's 360 Watts.  You don't need to worry about power management.  You have plenty to run everything at once.

diettelo, 2017-12-27:
so can i change it?
at the moment:
total 175w
buildplate 150w
Extruder 35w

so can i change extruder to 50w
and total 200w?

diettelo, 2017-12-27:
and is pid calib important again?

gr5, 2017-12-27:
Yes those changes are fine.  Since you don't need power management - just make all the power consumers add up to less than the power supply.  So if you set bed to 1 watt, nozzle to 1 watt and budget to 5 watts then power management will not curb/lower the power to anything.  Power management will be effectively disabled.

Again - setting budget to any value as long as it is larger than all the other items means you have more power than needed so no need to manage the power.

diettelo, 2017-12-27:
thanks i disabled power budget.

closed 2018-04-26 by TinkerGnome