TinkerGnome/Ultimaker2Marlin #85: 17.10.1

xile6, 2018-01-14:
Having a problem.
I loaded the firmware a few days ago and did a few prints.
But today i could not get it to print. It would heat up the bed and nozzle to correct temperature but then do nothing just sit there.

I factory restored and sams problem. On all gcodes.

Loaded 16.12.1 and everything is working correctly.

gr5, 2018-01-14:
This sounds like a hardware problem.  Please post this on the ultimaker forums and I'll give you some ideas and ask you some questions.

gr5, 2018-01-14:
Oh wait - you did factory reset but didn't go back to original firmware?  Try default firmware first and if it still has the problem then post on forums.  If original firmware works fine then let us know here.

xile6, 2018-01-14:
I have been on the tinker firmware for a while. Was one 16.12.1 last monday.
Upgraded to the 17.10.1 tuesday. (Jan 9th).
Clear eeprom
Did a few prints without problems last week.
Turn it on today to print and it just heated up the nozzle and bed. Move into the pruge loctations but never printed.

Aborted the print.
Turn off and back on. Same problem.
Factory reset. Same problem.
Try an old gcode file that i printed last week. Same problem.
Stays stuck on heatting never prints.

Uploaded the 16.12 firmware.
Try to print the file and its back to printing fine.

I will reupate to the 17.10 again and see if it happens again.

gr5, 2018-01-15:
Try turning off "geek mode" in the tinker marlin menu system.

xile6, 2018-01-15:
Same problem.
Ultimaker 2 extended, converted to dual
Upgraded because of the scrollsable filenames.

Running 16.12 print a file just fine.
Uploaded the 17.10 try to print the sams file an it stays on heating 100%.
Clear eprom (m502, then m500) same problem.
Factory reset and same problem.
Disable geek mode and progress bar stays at the end for heatting. Check the temps in tune and both hotend and bed are at temps. 205/205c 60/60c.

I was reading and seems that other have had the no printing problems with dual setups and 17.

Is there an easy way to just add scrollable filenames to 16.12?

TinkerGnome, 2018-01-20:
@xile6 Since you mentioned the dual variant:
Is there - by any chance - a temperature command for the unused nozzle in the start gcode? Something like "M109 S0 T1"? If so, remove it and try again.
Or if you still use UltiGcode: do you have selected a material for the second extruder (on the printer)? That's not optional.

xile6, 2018-01-20:
Same files. Just different firmware.
No T1 codes for my single color prints.
And yes both extruder have pla and it is setup under material for each extruder.

Not sure what the problem is.
When i factory reset and load 17.10.1
It simply wont print.
Ill try a few more thingd if needed. But i was really only upgrading from 16.12. For the scrollerable file names

gr5, 2018-01-21:
@xile6 I had these exact problems when printing through USB (heats up and then gets stuck).  But you are printing from SD card, right?

xile6, 2018-01-21:
Yes off sd card. Thats the only way i print

TinkerGnome, 2018-01-21:
@xile6  - can you provide a short (sample) gcode-file which shows the problem? I would like to try to reproduce it.
(only if possible, upload it somewhere and share the link)

xile6, 2018-01-24:
Sorry about the late reply.

dropbox to gcode
https://www.dropbox.com/s/7u2y8ghmcwcj1mr/UM2EX_bioshock_eve.gcode?dl=0

but it happen will all gcodes that i printed just fine on the 16 firmware