TinkerGnome/Ultimaker2Marlin #87: How invert X,Y,Z

Alekmarcus, 2018-02-06:


Alekmarcus, 2018-02-06:
..at Source code ? I have Chinese (clone) UM2.    I can`t change it by menu because start first wizard setup.  In source  code  line is -  #define INVERT_Z_DIR  (axis_direction &  4)

TinkerGnome, 2018-02-11:
You can change the "DEFAULT_AXIS_DIR" to your needs (just a few lines above your quoted one). The invert-flag for each axis is represented by a bit in the variable (axis_direction). So - flip the according bit and change the predefined DEFAULT_AXIS_DIR accordingly.

Alternatively: skip the first run wizard (*) and use the menu.
(*) - that means: change or remove the condition here:
https://github.com/TinkerGnome/Ultimaker2Marlin/blob/geek_mode/Marlin/UltiLCD2.cpp#L218

...or go the hardware route and swap one pair of the stepper motor cables to invert the rotation direction.

closed 2018-04-26 by TinkerGnome