TinkerGnome/Ultimaker2Marlin #88: Feature request: scroll title in screen (UM2)

UM2user, 2018-03-29:
I cannot believe that nobody has come up with this before but I cannot seem to find any news on this..
Since the release of the UM2 I asked myself why such a simple thing as scrolling a design title in the display was not possible. Is this so hard to fix. or has there not been anyone who sees the advantage of this?

I have multiple designs in different filament types and versions so i have come up with this format:
[image]
But this results in a cut of title name, so it is not always clear which file I need to print.

Any thoughts?

Thank you.

foreachthing, 2018-03-29:
here: https://github.com/TinkerGnome/Ultimaker2Marlin/pull/16
On my UM2, it scrolls the title of the selected file. Using 17.10.

UM2user, 2018-03-29:
Crap.... for how long is it there??? just upgraded this morning from 16.08.2 to 17.10. 
found it... works like a charm. 
Thank you.

foreachthing, 2018-03-29:
Glad I could help.
To answer your question: TinkerGnome committed on Feb 18, 2017. So, it only has been here for about a year ;-)

UM2user, 2018-03-29:
haha well I guess it wasn't present on 16.08.2 so upgrading was worth it only for this thing ;-)

thanks.

closed 2018-03-29 by UM2user