TinkerGnome/Ultimaker2Marlin #90: Compile Mark2 firmware?

yyh1002, 2018-06-13:
Hi, 
I'm trying to change the temperature difference limit from 25 to 100.
Downloaded the source files from mark2 branch, edited and uploaded it. 
But the machine recognises itself as standard UM2+ with default bed leveling points and no dual head options.
Do I need to make some changes in the source file before compiling for Mark2?

Thanks

gr5, 2018-06-13:
You have to build using package.sh which sets some crucial "macros" (it overrides some of the settings in configuration.h.  More info here:
https://community.ultimaker.com/topic/16856-how-to-build-marlin-the-way-tinkerghome-and-ultimaker-do-it-if-you-have-windows/

yyh1002, 2018-06-27:
@gr5 Thanks for making the instruction!