TinkerGnome/Ultimaker2Marlin #91: UM2+ Single Extruder 17.10.1 nozzle target 0

mlizbeth, 2018-07-02:
I've looked through the issues but can't seem to find anyone with the exact problem.

This happens with Simplify3D, Cura, and even trying to set EEPROM settings on the printer itself. Slicer overrides don't work and I always have to set my nozzle temperatures manually.
In S3D I am using standard gcode with Marlin FW settings. In Cura I am using UM2 gcode.

Here are the gcode settings from S3D https://pastebin.com/Qx3H021A

I will dig through the source in the meantime, though I am not sure at what update this bug occured.
To clarify, everything else is working nearly flawlessly, just need to fine tune extrusion settings!

gr5, 2018-07-02:
Normally temperature is set on the UM2 and not in Cura.  And there are 4 temperatures - one for each of the 4 nozzle types.  Setting this works best by creating the material.txt file editing it and reloading.  If you want to set temperature in cura you have to change gcode flavor to "Marlin".  Suddenly this will enable temperature settings in Cura.  

Other settings that on the UM2 are in the firmware and not controlled by Cura: bed temp, retraction distance, retraction speed, filament diameter.  So if you switch to Marlin mode double check that those values in Cura are all correct.

mlizbeth, 2018-07-02:
I tried auto calibration and manually setting the temperatures on the UM itself, but it doesn't "stick" or write it to EEPROM I suppose. When I go to print a file, the bed is set to on, but the nozzle is always set to OFF/0 degrees. 

What about simplify 3d? When I print a gcode from that (with marlin gcode) it says "this slicer will override defaults", sets bed to 60C (as I want) but nozzle stays at 0 until I manually tick up the values myself.

gr5, 2018-07-03:
Go into the material menu and select the option to write to materials.txt on the SD card.  Then bring that over to a computer and edit materials.txt with a text editor.  Be very careful not to introduce any syntax errors (don't add any spaces or comments or commas or remove any.  Just change the numbers only).  Then put the SD card back in the printer and read the materials.txt file back into the UM2.

gr5, 2018-07-03:
I don't know what the right mcodes are to set the temp on UM2 but you could choose "marlin" output in cura and open up the created gcode file.

mlizbeth, 2018-07-03:
I *believe* it's working as intended. Does the nozzle not turn on when printing until the build plate is done heating? 
However it won't turn on in preheat unless I set it myself. Is this also intended?

gr5, 2018-07-03:
Ultimaker likes to heat the buildplate first and then the nozzle.  Or at least heat it most of the way.  This is because some materials (particularly ABS and PVA) cook slowly when at normal printing temperatures and can caramelize into a nozzle clog.  PLA doesn't really have this problem.  I suppose it also saves power and also the bed uses a lot more power when cold so UM likes to avoid tripping the power limit on the power brick (is a problem for some users but rare).

Anyway the point is, typically you have to wait for the build plate to heat up before the nozzle heats up.  This is true in "ultigcode" format gcodes.  But in reprap gcodes or marlin gcodes this should be done with gcodes and the UM2 should follow the instructions in the gcodes.  For cura, I believe it creates gcodes that heat the bed first.  Not certain.  I'd be surprised if S3D heated the bed first.

mlizbeth, 2018-07-03:
Here are the gcodes from S3D and Cura, up until it begins printing. 
What is better, absolute or relative extrusions?

S3D: https://pastebin.com/1BF2afsZ
Cura: https://pastebin.com/nrjU6dHb

gr5, 2018-07-03:
Yeah well:
M140 is set bed temp and immediately run next gcode
M190 set bed temp and don't do next gcode until bed reaches this temperature (aka set temp and wait)
M104 set nozzle temp and immediately run next gcode
M109 set nozzle temp and wait until nozzle reaches this temp until running the next gcode

So looking at S3D it doesn't heat the nozzle until bed is at full temp.  The code is silly with the redundant (extra) M140 and M104 but it works - it doesn't start heating nozzle until bed is up to temp.

Looking at Cura it does the same thing.  You can change this if you want.  There are many ways to do this - for example you could heat bed to 45C with M190 (so it waits) then set bed *and* nozzle to 60C/190C but don't wait so they both start heating at the same time once bed reaches 45C, then set them to 60C/190C again but wait until they reach those temps.  This will speed things up a bit because it starts the nozzle heating when bed reaches 45C.  Or you can just start both going at the same time and wait until both are at full temp before printing.  Lots of choices.

So I don't think there are any bugs in tinkerMarlin.  You just have to be patient.  Or change the cura start codes.  Or the s3d start codes.

gr5, 2018-07-03:
Closing this issue for now.  message me directly if you still think there is a bug and I'll reopen.  Or contact me on the Ultimaker forums (gr5).  Or email me at my store:  thegr5store@gmail.com

closed 2018-07-03 by gr5