TinkerGnome/Ultimaker2Marlin #94: 1.75mm version

joshoowa5, 2018-11-16:
Please create a version with 1.75mm filament diameter preset.

foreachthing, 2018-11-16:
Go to `Filament Settings` - `Filament` - `Diameter`, then change it to 1.75 mm (measure the actual filament for more accuracy).
After that, save the Profile - DONE!

gr5, 2018-11-16:
When saving the material on the UM2 it's not obvious but at the end you have to choose PLA to save it back to the PLA profile.  It will then remember that forever.  When you upgrade to other versions of Marlin it should remember the old profile settings that you saved.  So having Marlin default to 1.75 won't help much if you have ever installed Marlin before.  You would have to update marlin and then do "reset to factory settings" to get the 1.75mm material settings.