TinkerGnome/Ultimaker2Marlin #95: 17.10.1

diettelo, 2019-02-05:
When printing the final layer with small area and or a combination of very often retraction. Nozzle stucks at the End and Filament is backwarded to the Extruder. After that Buildplate is going down and Printhead is homong.
Printed at 60 mm/sec

gr5, 2019-02-05:
There is a bug where if you set flow rate to something greater than the default (100%) at the end of the print it will retract by all that amount that the flow was set to.  So if the entire print is 2 meters of filament and you had flow at 110% then it will back off by 2 meters X 10% or 20cm.

Similarly if flow is less than 100% it will overextrude at the end.

While the extruder is moving it takes a while and the nozzle is "ruining" a spot on the print.

diettelo, 2019-02-05:
Hmm thats funny because i checked files and it was set on Simplify3d on 1.00 (100%)

gr5, 2019-02-06:
Please provide a link to the gcode created then.  It could be something that cura doesn't create but that S3D does.  Something perfectly reasonable but something that only S3D does.