TinkerGnome/Ultimaker2Marlin #99: Ultimaker 2 + with E3d

nvcheeko, 2019-05-01:
Hi Guys

Anyone ran this firmware with E3D head just wondering if i need to change anything in the firmware it will be ok 

Cheers

gr5, 2019-05-01:
This firmware works only with PT100 temp sensors. PT100 sensors are fantastic because they can take very high temperatures and they all have the same restance/temperature curve. More common (and probably E3D has this) are thermistors e.g. 100K thermistor or 10K thermistor.  No two thermistor brands have the same resistance/temp curve.

So make sure the E3D head (and they have many different heads) has a PT100 or alternatively see if you can buy a PT100 cartridge that fits the E3D head.

Also this works with heaters that can take 24V.  Some heads have heaters that are designed to work with 12V only.  If you put 24V into a 30Watt 12V heater it will put out 4X power or 120 Watts and likely melt in a few seconds.

nvcheeko, 2019-05-01:
Sweet yes ill definitely use PT100 they recommended it.. cool so no firmware change.  Thanks for your help.

Regards 
Peter

nvcheeko, 2019-06-05:
You can close this ticket, Thanks Gr5 for your assistance

closed 2019-06-05 by nvcheeko