TinkerGnome/Ultimaker2Marlin #101: Ultimaker 2 w/ FW V19.03.1 - Head pause while printing w/ buzzing sound

bitsandnumbers, 2019-05-18:
Hello,

First, thank you for this work which give a new breath to my old U2 :-)

I'm very happy with this firmware right now, if not for this annoying issue that scare me more than doing any arm to my prints.

So basically, the head halt for like 10 seconds while printing and make some kind of buzzing sound, and then resume printing like nothing happened. I'm always afraid that filament would burn and get stuck in the nozzle, but until now it never happened. Still, it's quite annoying and could fail a print someday. It's also not reassuring and I wonder if it does any arm to the printer itself...

I don't know if it happened to someone else, since I didn't find any mention of this issue online. For me, it happens regardless of prints and settings. It just seems random.

TECNOTOUR4343, 2020-03-30:
Hello bitsandnumbers, 
I have installed this firmware too and I really love it. 
I also have seen once the problem you are describing, the head gets stuck for over ten never ending seconds and you can hear a sharp noise, then it continues to print as if nothing happened.

frez-knee, 2021-01-13:
I have a similar problem where the printhead will stop randomly during a print, however, it will full unload and then reload the filament and extrude a fair amount before resuming. This obviously leaves an unsightly blob on the side of the print. I've noticed that it happens on the outer shell mainly and if I recall correctly, while printing from SD and Octoprint. It doesn't happen very often, I haven't been able to find exactly what causes it.

gr5, 2021-01-13:
I've seen this on most of my UM2 printers - I've never printed through USB so it's only been through printing from SD card.  What happens is it's getting read errors.  Sometimes for example it might say G1 X123.45  but it gets randomly change to F1 X923.45 and then it gets "out of bounds" error.

Sometimes it's the E axis so instead of say E1234.56 it gets E1034.56 so if you look carefully it will retract 200mm (about 1/3 of the bowden) and then on the next move it will properly move back to where it's supposed to be.  All the while aparently "pausing" if you've been listening to the printer.

The solution is usually to clean the SD card reader slot.  The first time this happened I actually found a hair in there.  I took it all apart but usually compressed air is enough to clean it out pretty well.

gr5, 2021-01-13:
Once it happens once or twice per hour of printing it's much too excessive and time to repair that printer.  But all my printers have done it occasionally - e.g. once every 6 months or so that I notice.

frez-knee, 2021-01-13:
@gr5 I've seen you mention the SD card reader issue on other similar issues. What's interesting is I've yet to see this occur on my other UM2+, converted to Mark2 that's running the same version. Only happens on my UM2+, single extruder with Bondtech Feeder. 

The screen was asleep when I caught it happening, so I didn't see any pausing notification on the screen. But I did watch it completely retract the filament the full length of the bowden (720mm) plus some extra as I could still hear the extruder gear spinning a brief moment. Then it must have loaded that same amount (maybe 800mm?) since it extruded a blob on the print, like I mentioned earlier. 

I can't recall exactly when the last time it did this, was probably a few months ago. I didn't see anything obvious in the slot, but I did clean both printers out with some compressed air just to be safe.

gr5, 2021-01-13:
It's not a pause.  It's a pause in the "song" of 3d printing but it's not a pause - it's a massive retract.

The problem is that bytes change.  And it happens more often than you think.  For every obvious 200mm (or larger) retract, there are subtle ones where the print head moves outside of the print for no reason and then goes back in.  Random Z moves are are not a problem because only one in maybe 1000 moves are a Z move and this problem only happens to something like one in 100,000 moves.  something like that.  Or other changes where the gcode isn't recognized and just skipped over.  This can create invisible "missing infill line" or a little more visible tiny bump due to head going directly diagonal to two points later or it can create a droopy string on the side of the print because the head moved outside of the print for a second while still extruding (but extruding slower than usual for such a long move).  (droopy strings look different than normal stringing - normal stringing tends to have horizontal strings).

But realize for every "long pause" there is one of these other errors that are much harder to notice.  Here's the format of the most common gcode:
G1 X123.454556 Y56.77123 E1023.72634

If you think about what happens if you change a single bit it means you change one of the above characters.  If you change the "G" then the entire gcode is ignored.  If you change the X then it does the gcode fine except it doesn't move the X at all (but recovers on the next gcode).  And so on...

Yeah - cleaning might not be the proper fix.  I tried lots of things.  Changing SD cards.  changing SD brands.  Playing with the 2 ribbon cables that connect SD card reader to the arduino.  Swapping SD card reader circuit boards among to UM2 printers (that was inconclusive).

Intermittent problems are hard to fix.  Anyway the cleaning alone definitely lowered the incidence from once every 5 minute print to once every 10 5 minute prints (was printing stretchlets).  And further fiddling got it to once every 100 prints or so such that I don't notice the problem very often.  The cleaning made a 10X improvement for sure.  It was very obvious.  It was a thorough clean where I took the PCB out (the little one with the SD slot) and shoved implements into it and used isopropyl alcohol and so on.  It's possible the jostling of the flat ribbons was the fix - they aren't shielded.  Supposedly they should stay a few mm apart and not touch each other for long (>1 inch) lengths.  There's a phenomenon called "cross talk" that happens when two wires are touching for many inches.  Digital signals can hop wires.  Putting foam between them could help - space them more than 2mm apart.

frez-knee, 2021-01-13:
Thanks for the info and explanation. 

Since maybe 6-8 months of running the Tinker FW on my single extruder UM2+, the large retract and unretract have happened twice (that I've seen). I can't recall if I've noticed the smaller more subtle ones like you mentioned. 

FWIW, the Mark2 has a 4GB SanDisk and the UM2+ has a 8GB Verbatim (the Verbatim came with the printer from Ultimaker). 

Just thinking out loud, is it possible that Gcode flavor could have any effect? I can't remember what flavor I used the last time it happened, but this time it was UM2 flavor, not Marlin. I usually swap back and forth between flavors depending on if I'm printing from SD card or Octoprint. I just like UM2 flavor for being able to save material settings on the machine... 

Interesting point regarding the cross talk, this is why the stepper motor wires are twisted pairs. It's meant to mitigate that exact issue. As well as the ferrite cores, if I remember correctly.

gr5, 2021-01-13:
The problem is frustratingly difficult to diagnose.  I kept notes and the one key thing was cleaning out the SD card slot.  I know for sure there are bad bits.  During much of the experimenting I was printing the exact same 3 minute part on the exact same printer with the exact same SD card.  I took the SD card out and did a "diff" on it versus the original file and it was fine.

So the problem is live - while reading the SD card - sometimes a bit gets flipped.  Always only one bit (e.g. a zero changes to a 1,2,4, or 8 but not 5).  It could be the SD card itself, the contacts, the electronics, the wiring, something in the arduino - it's easy to "fix" when it happens at least once every 3 minutes.  Much harder when it's once every 3 hours.  Cleaning was the thing that made it go from 3 minutes to 10 hours.  I can't get beyond that as experiments now take weeks to test just one variable.  I've seen this problem on all 4 UM2 printers at one time or another but pretty rare.

gr5, 2021-01-13:
Before the cleaning I tried moving the ribbon cables, changing SD cards.  The cleaning was the first thing that made a difference.

gr5, 2021-01-13:
The contacts were quite brown - using isopropyl alcohol got them back to shiny gold color.

frez-knee, 2021-01-13:
On the 4 UM2 printers you've seen exhibit this problem, were they all running Tinker FW, or have you seen it happen on stock FW as well?

gr5, 2021-01-13:
Stock FW as well.

frez-knee, 2021-02-27:
I'm not sure whether to continue this conversation here or not, but I thought that I would mention that I started noticing an issue with the SD card reader on my UM Mark2 also. However, the issue seems to manifest itself as a layer shift and not a huge retraction like I was getting on the UM2. 

Perhaps it is another case of bytes changing like you mentioned earlier, but it's affecting the XY position and not the E position.

It also started happening very frequently out of nowhere. I hadn't changed anything in my workflow, other than updating Cura to 4.8. But prints come out fine via USB. It would happen multiple times within the span of 30~ minutes. It also would make this loud noise/vibration as the printhead rapidly moved back and forth in random directions. 

I tried cleaning the contacts on the SD card with IPA and blowing out the slot with air. That seems to have helped, now it happens every few hours. Still not reliable enough to print with SD, though. 

What would be the repair for this issue? Replace the card reader on the printer? I believe you said that was inconclusive.

gr5, 2021-02-27:
I'm not sure.  I just don't have enough data.  And the problem is intermittent so it's really hard to know what works as you might have to print for a few weeks to be sure it's fixed.

For sure, cleaning the contacts improved things but not 100%.  I took it apart, took the board out, used long skinny tools and isopropyl alcohol.

It's possible that the 2 ribbon cables need to be spaced out more.  Even 1mm farther apart might make a difference.  There is something called "cross talk" in electronics when you have digital signals going down ribbon cables.  The signals if parallel and very close together for long distances have stronger electromagnetic coupling such that a bit from one wire can mess with a bit in the other.  I HAVE NO IDEA IF THIS IS A PROBLEM but I tried to space the 2 cables apart a little more.  Even just a few mm might be enough as the forces drop off by the square of the distance I think?

And of course it could be your SD card as well so try a different one.

Yeah I swapped cards between 2 UM2s and the problem didn't move with the cards.  Then years went by and I forget what I did.  I saw the problem twice in the last 10 months.  It's really rare now.  I need to take really good notes to diagnose it fully.