TinkerGnome/Ultimaker2Marlin #102: Running the Ultimaker Extended plus

nvcheeko, 2019-06-03:
Hi I recently upgraded my firmware to UM2+extended as i removed all the Dual stuff and Tinker dual wouldn't work anymore due to removing the other PT100 had PT100 Error, so i upgraded to UM2+ extended  not Dual and now it says i have a X or Y end stop error.

I can download the Ultimaker firmware from Cura and works and prints fine.

Anyone have any ideas

gr5, 2019-06-03:
I assume you have an extended, right?  It's taller than a normal UM2?

After downgrading firmware or cross-grading (is that a word?) you need to do a FACTORY RESET.

Each new version of the firmware knows how to upgrade from older versions but they don't know how to upgrade from newer versions that didn't exist yet at the time.  I could explain in more detail but whenever you downgrade firmware you need to do a FACTORY RESET.  Or if you upgrade from tinker version to UM version that can also mess up a lot of settings that are fixed with FACTORY RESET.

nvcheeko, 2019-06-04:
I will give it a go tonight, i will reset the firmware after uploading the Tinker gnome UM2 Extended + I was printing fine on Dual but im getting Flex drive so thought i would remove everything as Dual is no purpose for me. Ill keep you updated. Standard is no fun when printing from USB

gr5, 2019-06-04:
>reset the firmware
It should be in the menus somewhere: "FACTORY RESET".

nvcheeko, 2019-06-04:
Worked a treat thanks heaps,

closed 2019-06-05 by gr5