TinkerGnome/Ultimaker2Marlin #103: Pid Tune 50W

diettelo, 2019-09-14:
I cant get Pid Tune my Nozzle 
Got 50W but i goes over the 15 degree
Autotune failed.
Can you increase the amount?
Happens even on low temp

gr5, 2019-09-14:
Suggested PID values for 50W heater on UM2:
P,I,D
5, 1.2, 50      -- basically the 25W values cut in half
5.6, 0.35, 23  (neotko recommendations X35/50)

Either of the above values should work quite well.  You can be off by quite a bit from ideal and still get great results from PID controller.

gr5, 2019-09-14:
To keep autotune from failing you could tune with fan on 100% and nozzle close to the glass bed.  Or you could tell autotune to tune on a higher temp like say 260C but this second option is a worse idea and not as realistic to normal printing conditions.

diettelo, 2019-10-19:
How can i go close to the bed? With menu only or do i need a computer?
Do you have M codes for me?

gr5, 2019-10-19:
Please post questions like this on the ultimaker forum.  This github area is for bugs with tinkermarlin.  Here's the forum:
https://community.ultimaker.com/forum/106-ultimaker-3d-printers/