TinkerGnome/Ultimaker2Marlin #107: Large extrusion at end of print with pause & change plugin

Sisko4, 2020-11-14:
Using an UM2+ and TG 17.10.1

When i setup a pause with the plugin in cura, change the filament and start again, i get an extrusion instead of a retract at the end of a print, wich produces a big button if filament on the print.

Don´t happen with a manual pause.

Since then i don´t want to try a programmed pause anymore.

Does someone have this problem too?

gr5, 2020-11-14:
I have this problem if I set flow rate to anything other than 100%.  I think if you set flow rate to a value less than 100% it over extrudes at the very end and if you set the flow rate to a value more than 100% it under extrudes at the very end.  I think it's a bug with how Cura keeps track of the position of the E axis throughout the print.  At the very end it wants to do a retract of the extruder - 3mm or so.  I don't remember if it switches the E axis to relative mode or if it uses the G92 command (set position) but either way this is something that doesn't happen for 99% of the print and happens at the very end.

I think it's multiplying some moves by the flow factor to get the actual E position but then when you do this 3mm retraction it thinks the extruder is at one position but is at another because of the flow %.

Changing flow in Cura will not affect this because cura adjusts all the E values (extruder positions) in the gcode and the printer is at 100% flow.

Did you change your flow value (on the printer, not in cura)?  If so your solution is to change it in cura instead.  Another solution would be to get rid of the last bit of gcode that does a retraction before homing the printer.

gr5, 2020-11-14:
If you didn't change flow then I think it's a similar but different bug where Marlin is storing 2 different E values (or some kind of E offset).  And one of these gets messed up.

Sisko4, 2020-11-14:
Never touched the flow rate to be honest, i never got any problems with extrusion itself.
I will try to reproduce this problem on a test cube, if it happens again, i will do it manually in the future.

Sisko4, 2020-11-15:
Btw: what exactly does the geek mode do when it's on?
I can't find an explanation for it.

gr5, 2020-11-15:
Try it.  You just get a lot more in the GUI when printing.  So for example it shows nozzle temp, fan speed, bed temp, speed, flow and like 5 other things all on one tiny display.  I forget the rest - maybe Z position?  And you can use the scroll wheel to settle on any of these and there is more info and the ability to instantly change that one thing (like fan speed).

I love that when you change the % speed it tells you the speed the printer is moving at that very instant (so I don't have to do the multiplication in my head).  I love a lot of other things.  But there is some thought out there that geek mode makes the printer more likely to crash.  I have all 4 of my UM2 printers in geek mode and I have printed many hundreds of things with no crashes the last 6 months.

gr5, 2020-11-15:
>Never touched the flow rate to be honest, i never got any problems with extrusion itself.
Okay well I still think the bugs are related.  When you pause the print it has to retract and then when you continue it unretracts and I think there is some extruder-offset component in Marlin that gets messed up for when the print is at the end.

I print torque wrenches that are very fickle - my goal is 0.9 Nm of torque (+/- 0.2 Nm).  If I increase flow by 1% that changes the torque by about 0.1Nm.  And vice versa.  I typically do non-stop wrenches for a day on the same printer (40 minutes each).  Testing and labelling each wrench as I go and adjusting flow after the first wrench.  But I'd get these huge blobs at the end of the print if flow was < 100%.  So now I adjust with temperature - 5C changes the wrench by about 0.1Nm.  I haven't had a blob at the end of my prints since I used temperature instead.

Note that changing filament color, or brand changes the torque a bit.  When I get near the end of the spool it underextrudes more and I have to raise the nozzle temp to compensate.

gr5, 2020-11-15:
I mean think about it.  When you set flow to 90%, and you are at position E1000.0  (one meter into the print).  And you now tell the printer to print a line with 1mm of filament (E1001) it really needs to move to E1000.9.  But then when it wants to print the next mm of filament on the next line and the gcode has E1002, the printer now needs to move to E1001.8.  So the printer is keeping track of 2 positions at the same time where the gcode thinks it is and where the printer thinks it is.  Something like that?  I'm just guessing that this is how some programmer implemented it.  So there is either two variables that store "E" or there is an offset.  Or something.  Somehow this gets messed up when you switch to "relative" mode and do the final retraction.

Something similar could be happening when you do pause print.

Maybe I'm reaching at straws but I've DEFINITELY seen this bug and it went away when I stopped adjusting flow.