TinkerGnome/Ultimaker2Marlin #108: 50W heater cartridge - changes needed?

Sisko4, 2020-11-18:
I maybe add a 50W heater to my UM2+ in the future, do i need to change something in the firmware to use it or is it just plug & play?

gr5, 2020-11-18:
There are two things that you might want to change.  PID and power budget.

The power budget is pretty generous on the UM2 so it might be fine.  You know it's not fine if the power brick cuts out for a fraction of a second and your printer reboots in the middle of a print and goes back to the start screen (no error, it just stops and if you don't look at it within a few seconds it's already sitting at the main menu screen).

So I would recommend tinker marlin version 19.X for just this reason but really it's probably fine if you don't bother.  Tell it the bed is 150W, heater is 50W and budget for these 2 devices is about 180W.

PID - if you don't adjust this then it will overshoot quite a bit if you are heating to temps below 260C.  But it should equalize to the right temp within a few minutes so not a big deal if you use values meant for 25W heater.  Or it might oscillate in temp.  It's easy enough to just start a print and watch the temp for 5 minutes to see how much it varies.  You can set PID easily in the tinkergnome menus.  Or you can do it on UM2 original firmware with gcode commands.  My recommendations:
25W PID 10, 2.5, 100
35W PID 8, .5, 33 (neotko recommends)
40W PID 6.5, 1.5, 60
50w PID 5, 1.2, 50

Sisko4, 2020-11-19:
Thank you very much!
I just checked it and saw that the PID values in my printer were set for a 25w cartridge, but the um2+ has 35w.

I now know why i had some problems with the temperature, one error less i need to check!

Sisko4, 2020-11-21:
Are this values correct for the 35w heater?
35W PID 8, .5, 33 (neotko recommends)

I am curios that there is a 0.5 in the middle while others have a higher number.

gr5, 2020-11-22:
I know, right?  Well you can have multiple values of PID settings that are quite different and yet still work quite well.  I spent about 4 hours a few weeks ago playing with PID values and I could never get it as good as I wanted.  Partly because the PID doesn't even turn on until you are within some delta of the goal temperature.  That was a bad decision.  I think the delta is either 20 or 30C?

Anyway, if you use the wrong values by a factor of 5x or so it will probably still work just fine.  I'd trust neotko.  He's pretty smart. 
 He probably got those values by using the autotune feature built into marlin that calculates the PID values for you automatically.  So they are probably quite good values.

In theory you should be able to take the UM values and scale them down by the ratio of power.  So the values for 50W versus 25W heater - you should be able to just divide the 25W PID values by 2.  But in practice that doesn't work (I know - I've tried).  And I don't understand why not.  I understand exactly how PID is supposed to work.  But something is wrong.  Partly the problem is that it heats up much faster than it cools off and PID is expecting a more symmetrical response.  Partly the problem is that there is this delta.  But I don't think that explains it all.

Just try it and set the temp to 210C and see how much it overshoots and how long it takes to stabilize (or does the temp oscillate forever?).  I've found that doubling or halving any of the PID numbers does not make a big deal of difference.

Sisko4, 2020-11-22:
Pretty good explanation!
I got the standard ultimaker values for now and try the neotko values at a few test prints.
Have a few problems with my printer and got 3 faulty 3dsolex couplers in a row...
Hope after everything is done it works perfect like on day 1.

Tahnk you very much!