TinkerGnome/Ultimaker2Marlin #109: Error compiling V17.10.1 with Arduino 1.8.13 for UM2 Extended with E3D V6 and Titan extruder

Andrei-2k, 2020-11-19:
I changed only 2 files and this is probably a noob error but am hoping someone can help?

`Arduino: 1.8.13 (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\Marlin_main.cpp:47:0:

sketch\machinesettings.h: In member function 'bool MachineSettings::has_saved_settings(uint8_t) const':

machinesettings.h:24:87: error: operands to ?: have different types 'MachineSettings::t_machinesettings*' and 'bool'

  bool has_saved_settings(uint8_t index) const { return (index < MAX_MACHINE_SETTINGS) ? settings[index] : false; }

                                                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~

In file included from sketch\UltiLCD2_menu_first_run.cpp:9:0:

sketch\machinesettings.h: In member function 'bool MachineSettings::has_saved_settings(uint8_t) const':

machinesettings.h:24:87: error: operands to ?: have different types 'MachineSettings::t_machinesettings*' and 'bool'

  bool has_saved_settings(uint8_t index) const { return (index < MAX_MACHINE_SETTINGS) ? settings[index] : false; }

                                                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~

In file included from sketch\UltiLCD2_menu_material.cpp:8:0:

sketch\machinesettings.h: In member function 'bool MachineSettings::has_saved_settings(uint8_t) const':

machinesettings.h:24:87: error: operands to ?: have different types 'MachineSettings::t_machinesettings*' and 'bool'

  bool has_saved_settings(uint8_t index) const { return (index < MAX_MACHINE_SETTINGS) ? settings[index] : false; }

                                                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~

In file included from sketch\UltiLCD2_menu_main.cpp:14:0:

sketch\machinesettings.h: In member function 'bool MachineSettings::has_saved_settings(uint8_t) const':

machinesettings.h:24:87: error: operands to ?: have different types 'MachineSettings::t_machinesettings*' and 'bool'

  bool has_saved_settings(uint8_t index) const { return (index < MAX_MACHINE_SETTINGS) ? settings[index] : false; }

                                                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~

In file included from sketch\UltiLCD2_menu_maintenance.cpp:14:0:

sketch\machinesettings.h: In member function 'bool MachineSettings::has_saved_settings(uint8_t) const':

machinesettings.h:24:87: error: operands to ?: have different types 'MachineSettings::t_machinesettings*' and 'bool'

  bool has_saved_settings(uint8_t index) const { return (index < MAX_MACHINE_SETTINGS) ? settings[index] : false; }

                                                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~

exit status 1

operands to ?: have different types 'MachineSettings::t_machinesettings*' and 'bool'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

Andrei-2k, 2020-11-19:
I actually found a post that says to ad '!=0' and then the compile would finish but no hex was output but I also resolved that by making sure my code was on a local drive instead of network. A noob issue to seasoned developers I am sure. I will report back if firmware loads properly.

gr5, 2020-11-19:
First of all the compiled code is already available here: https://github.com/TinkerGnome/Ultimaker2Marlin/releases

So there should be no need to compile this yourself.  If you do wan to compile your own custom changes then here are instructions: https://community.ultimaker.com/topic/16856-how-to-build-marlin-the-way-tinkerghome-and-ultimaker-do-it-if-you-have-windows/

gr5, 2020-11-19:
The "!=0" edit is a good one.  @TinkerGnome take note.  I guess he is using a different compiler and this is a test for null pointers in an array called settings[] evidently.  The compiler is upset that settings[N] is a pointer and not a boolean.

Andrei-2k, 2020-11-19:
I was able to get a hex and upload it to the printer! a bootloader hex file also was output but I assume I ignore that one?

This was output but I hope I can ignore this? `

sketch\tinkergnome.cpp: In function 'stopMove':

sketch\tinkergnome.cpp:2113:47: warning: iteration 3 invokes undefined behavior [-Waggressive-loop-optimizations]

         TARGET_POS(i) = current_position[i] = constrain(st_get_position(i)/axis_steps_per_unit[i], min_pos[i], max_pos[i]);

                                               ^

sketch\tinkergnome.cpp:2111:5: note: containing loop

     for (uint8_t i=0; i<NUM_AXIS; ++i)`

The reason I didn't use one released is because I had to modify heat, max, stepper, etc( for the E3D kit with[ V6 hotend](https://e3d-online.dozuki.com/Guide/V6+Marlin+Configuration/5?lang=en) and [Titan extruder](https://e3d-online.dozuki.com/Guide/Titan+Marlin+Configuration/35?lang=en) that no longer has support that I installed on the printer 2 years ago and have not been able to use it since)

I see Tinker 17.10 DEV show up on boot so that's good, along with left and right menu items.

Now the issue is that within 2 min or rebooting I get a "ERROR - STOPPED Heater error Go to ultimaker.com/er03"  so I assume this has something to do with the fact that for this hotend the max temp is 285?

gr5, 2020-11-19:
> The reason I didn't use one released is because I had to modify heat, max, stepper, etc( for the E3D kit with V6 hotend and Titan extruder that no longer has support that I installed on the printer 2 years ago and have not been able to use it since)

Got it!

>  "ERROR - STOPPED Heater error Go to ultimaker.com/er03" 
That means the heater didn't heat up fast enough.  If the goal temperature is higher than the current temperature and PID is asking for full power (e.g. if the goal temp is 20 degrees higher than current temp) and if the heater does not increase in temp by at least a tiny amount for 30 seconds then you get heater error.

Sometimes it means the heater isn't working at all - or the heater cable is loose and intermittent.

Sometimes it means you are asking for too high of a temperature and the heater just doesn't have enough wattage to reach that temp.  I've gotten it with a 20W heater when printing with a large (0.8mm) nozzle at high speeds (0.4mm layer height) on a high temp material (CPE).  But that's pretty rare.  Usually it's a hardware issue like a loose wire.

gr5, 2020-11-19:
Oh and sometimes it means that the temp sensor isn't working.  The main reason this test was added to Marlin was to avoid having the printer catch fire possibly causing death simply because the temp sensor fell out or is defective.

Andrei-2k, 2020-11-19:
> Got it!
>
>
> That means the heater didn't heat up fast enough. If the goal temperature is higher than the current temperature and PID is asking for full power (e.g. if the goal temp is 20 degrees [...]

Thank you for mentioning this so I can watch out for it in the future. I think I resolved the issue, the guide that is available [here](https://e3d-online.dozuki.com/Guide/V6+Marlin+Configuration/5) says to change the temp sensor to `#define TEMP_SENSOR_0 5` which is incorrect I believe and only tried ignoring it once I saw the other "guide" [here](https://e3d-online.dozuki.com/Wiki/Ultimaker_Upgrade_Supplemental_Info) although I was worried I would burn something by not changing it initially.

I was able to get some prints going! My issues now are that I am still nervous to change temp to `#define HEATER_0_MAXTEMP 400` since I heard you can damage the thermistor I believe and probably have to look at installing an alternative with the V6. I also have to tune the feeder extruder and figure out why the latest Cura will not preheat prior to starting prints.

If you have any resource I can look to for optimizing start and end g-code for printer & Extruder I would love to get that right.

Also, I noticed you are also in Boston, as COVID tapers off I would love to buy you a beer (or coffee) since I also live in the greater Boston area. To stay in touch I am [here](https://www.linkedin.com/in/andreisavu/)

gr5, 2020-11-19:
I clicked "connect" in linked in.  Would be happy to go out for a beer or whatever this summer.

If you have the manufacturer and part number for your thermister (I assume you must have that - otherwise there's no way to select the right thermister table) then you can go to the datasheet for that part (google it).  That datasheet will tell you the allowed max temp where you won't damage the thermister.  For PT100 I think it's the melting point of platinum (insanely high).  Note that PT100's are all the same.  It doesn't matter who is the manufacturer, they all have the same table.  But a 100K or 10K thermistor has a different table from each manufacturer and some manufacturers have more than one type.  And the differences can be a lot - 50C temp difference for the same resistance.

Andrei-2k, 2020-11-19:
Well I left the settings alone because I think the e3D kit did not provide a new one so am using the UM2 default.

Thank you I won't go off-track anymore except to say that I used Arduino 1.6.10. When I tried 1.0.5 or earlier it didn't work (even though it says that's what Ultimaker uses)

closed 2020-11-19 by Andrei-2k