TinkerGnome/Ultimaker2Marlin #110: Different Offset each print UM2+

Sisko4, 2020-12-23:
I have a curious problem while printing the first layer, the distance between nozzle and printbed is not constant.
It varies on every print, one time it is perfect like i set it, other times it is almost to low and then higher.
I had this problem also with the original firmware this summer, but it somehow didn´t occur until a few weeks ago again.

The glas plate is new, all axes are oiled and the z-screw is freshly cleaned and greased.
I level perfectly each time, but it doesn´t help.
I did nothing change at the gcode or anything else, slicer is Cura 4.8.

gr5, 2020-12-23:
So I have seen this also and it is caused by a bit of plastic falling into the z homing switch.  If you lift up the bed and look at the bottom - you can see a long screw sticking down in the back.  Make sure it isn't loose.  Then look down and see where that screw goes in through the bottom of your printer.  clean all bits of plastic out of that hole.  Compressed air might work.  You should see only the shiny surface of the limit switch.

Sisko4, 2020-12-23:
Is checked, only a lot of dust behind the cover plate, but nothing that could cause the problem.
Switch and screw are okay.
Cleaned that area with compressed air too to be safe.

Sisko4, 2021-01-03:
Sad to say this, but i had to change to the custom Bondtech FW. The offset changed at every print start, now it works like it should. I have a stable offset at every first layer, something is wrong with the last stable release.