TinkerGnome/Ultimaker2Marlin #111: Ultimaker 2+ ERROR - Stopped X or Y switch broken

Fullmelon, 2021-01-28:
I am using an ultimaker s+ for work and i am trying to use the new tinker firmware. It doesn't matter which version i use i always get an error as soon as i start a print or home the head.
[image]
ERROR-STOPPED
X or Y switch broken
Go to: ultimaker.com/ER07

I don't have any problems printing with the original firmware but id like to use the tinker firmware because it's much more user friendly.

gr5, 2021-01-28:
This usually takes seconds to fix.  There are 2 rods that pass through the print head.  one goes left/right and passes *through* the left sliding block and is supposed to stick out well past the left sliding block.  Push the head to the back of the printer and that should hit a switch and hear the click.  Often the rod shifts over hundreds of hours of printing and doesn't stick out anymore.  That's the Y switch.

now push the head to the left as far as it goes until you hear the same faint click.  Both clicks (X and Y) are audible if you push the head around by hand.  That X switch is mounted with 2 screws through the top plate in the left rear corner.  the switch itself typically, gradually slides to the left over the years and can be loosened slightly and slid back to the right.  Alternatively, sometimes the print head hits the left side of the printer and scratches.  you can usually just bend the sheet metal back a few mm such that it no longer hits the left panel.  That's the X switch.

gr5, 2021-01-28:
Sometimes just bending the metal arm of the switch a tiny amount fixes the issue as well once you realize which switch it is.

vorobej72, 2021-02-08:
I am using Ultimaker 2 Extended with Tinker firmware Vers. 19.03.1
Unfortunately I have the same issue on my printer.
Have checked two rods for print head and to switches, bar X switch still don't work. Two switches works perfectly in the official firmware from Ultimaker.
Maybe during the first installation of Tinker firmware, the X switch was not activated when the printhead was not in home position?

vorobej72, 2021-02-08:
!!! SOLVED!!!
[image]
[image]
[image]

gr5, 2021-02-08:
You should do a FACTORY RESET.

Settings like these are stored in a special area (eeprom).  Each version of marlin stores the values in a different place.  Each new version of Marlin knows where the old variables are stored and can move them to the new locations but this broke at some point when UM Marlin and tinker Marlin split apart.

FACTORY RESET sets all these values back to their proper value.  You caught 1 error but there may be several others.

gr5, 2021-02-08:
@Fullmelon - take note!  You probably had the same issue.

vorobej72, 2021-02-08:
https://github.com/gr5
Thank You!