TinkerGnome/Ultimaker2Marlin #112: Random code jumps?

sapzxc, 2021-05-18:
Hi!

Ultimaker 2 extended.

I have glitches looks like code jumps. 
Prehistory: I bought used ultimaker and using it for a year. Sometimes I have glitches looks like gcode jumps. Today I tried to update "Ultimate Cura" and firmware. Aaaand I discovered that had Tinker firmware and it is not official firmware. I tried several times to print with official firmware but it is too hard without changing settings in real-time. Re-flashed my Ultimaker to last Tinker V19.03.1. And now I can not made printing stable. 
I tried generate code for Ultimaker 2 and for Marlin but no luck.

The problem shown in picture below:
[image]

Video of problem (sound on). Here example when left-down model was missed to print and strange behaviour of head looks same in each glitch.

https://user-images.githubusercontent.com/6615536/118712338-5b4ee880-b829-11eb-9bb1-778a6ecaa49d.mov

Please advice what I do wrong. 

Thanks.

gr5, 2021-05-19:
I'm not sure what is going on where you circled - those look somewhat normal if it hasn't finished the first layer.

However I watched the video and it looks like it's moving the extruder - A LOT - like maybe 300mm.  I've seen this on various UM2 printers - it's a hardware issue when reading the SD card.

I had this problem on one printer and fixed it mostly by cleaning the contacts in the uSD slot.  I took the board out and carefully cleaned all the contacts.  I found a small hair in there - I think an eyelash.

Afterwards it still did the problem but much less often - like once every 20 hours instead of once every 3 minutes.  It doesn't happen often enough for me to be able fix it further - it's just too rare and intermittent.

But the problem was definitely hardware - not software.

What happens is one character in the gcode gets changed so a move like:
G1 X100 Y100 E1234.45
changes to
G1 X100 Y100 E1534.45   (2 changes to 5) so it moves the extruder 300mm!  I think I can hear that happening on your print where it's retracting the filament half way down the bowden!

Then the next command will have the correct extrusion position so it un-retracts the same amount and continues printing just fine like nothing happened.

Other errors can have the print head move out of bounds and get an "out of bound" or "out of print area" error.  Or it can move the X or Y axis to the wrong position briefly and then recover and continue on the very next position.

By the way, when changing firmwares so much you probably should do "FACTORY RESET" on the menu system each time you load different firmware.  Otherwise you can get some weird behavior regarding certain settings like "prime position".

I have tinkerMarlin on all 4 of my UM2 printers and it works great.

sapzxc, 2021-05-19:
I tried 2 flash cards, bought another user control plate at aliexpress. This video right after factory reset.
And yes, you are right, sounds like extruder make maximum speed.
Ok, today will try one more another flash card and will review contacts. But why program do not show any errors?

sapzxc, 2021-05-19:
Contacts of previous control board with same glitches looks good, see image below
[image]

gr5, 2021-05-19:
That does look good.  Did you clean the gold contacts with isopropyl alcohol?

I don't know if this will help - but there is a common problem in electrical engineering called "cross talk" when signals are too close together along long wire paths.  I would put something (foam?) between the 2 ribbon cables to separate them slightly farther.  Even just 1mm farther apart might help.

If you already changed this circuit board then maybe it's time to change the other circuit board?

sapzxc, 2021-05-19:
I have changed main board and two flat cables between main board and control board. Now works normally (only 1st layer finished as for now). Thanks for advice. This was hardware problem. I think it was atmega issue, but not sure.

[image]

sapzxc, 2021-05-19:
No, problem is still here.
New main board. New wires. New SD card formatted in OSX in FAT. Factory reset. 
I will wait until printing finished and review gcode file. So here maybe 2 reasons: wrong gcode from Cura or errors in gcode reading/applying by firmware.

[image]
[image]
[image]

gr5, 2021-05-19:
Did you go into cura PREVIEW mode and slide vertical slider to bottom layer and then use the horizontal slider to see what order everything prints and make sure it prints the way you expect?

gr5, 2021-05-19:
I see 2 different problems.  
1) You showed in the video it doing strange thing (probably retraction) 
2) and also you seem to think it is skipping areas of the print - I don't think the "skipping areas of the print" is a printer problem - I think it's either normal or something that you should be able to see in Cura.

The changing of boards may have fixed problem #1.  I don't agree that problem #2 even exists or if it does exist it could be a slicing problem.

sapzxc, 2021-05-21:
Current research status below:

I made modifications in code to see what exact blocks we got in file reading process and what "status" of data stored in arrays when glitch happens. I paste my raw log output below.

Lets I introduce of my marks in log:
`"BLOCK"` - each block read inside function `SdVolume::cacheRawBlock`
`VOL_BLOCK_PREVPREV`, `VOL_BLOCK_PREV`, `VOL_BLOCK_CUR` - this 3 blocks stored in memory inside function `SdVolume::cacheRawBlock` right after reading each block with function `sdCard_->readBlock` and this 3 blocks shown when "Unknown command" error happens.

BLOCK(299E) - 299E it is address/number of block. Exactly 299E it is system block inside FAT table of my card that makes glitches offen. This block read each time when we need cluster information.

Below raw serial output and next after this log I will explain what happening here.

```
16:17:42.988 -> BLOCK(1308F):5 Y21.183
16:17:42.988 -> G0 X62.677 Y20.697
16:17:42.988 -> G1 F1800 X69.726 Y20.697 E40.21004
16:17:42.988 -> G1 X70.54 Y20.507 E40.22394
16:17:42.988 -> G1 X71.176 Y19.887 E40.23871
16:17:42.988 -> G1 X71.404 Y18.993 E40.25405
16:17:42.988 -> G1 X71.174 Y18.232 E40.26727
16:17:42.988 -> G1 X70.545 Y17.54 E40.28282
16:17:42.988 -> G1 X69.729 Y17.297 E40.29698
16:17:43.023 -> G1 X62.668 Y17.298 E40.41441
16:17:43.023 -> G1 X61.847 Y17.541 E40.42865
16:17:43.023 -> G1 X61.493 Y17.896 E40.43698
16:17:43.023 -> G1 X61.213 Y18.238 E40.44434
16:17:43.023 -> G1 X60.989 Y18.993 E40.45743
16:17:43.023 -> G1 X61.215 Y19.887 E40.47277
16:17:43.023 -> G1 X61.855 Y20.509 E40.48761
16:17:43.023 -> G1 X62.677 Y20.697 E40.50163
16:17:43.023 -> G0 F3600 X62.877 Y20.697
16:17:43.023 -> G0 X62.877 Y21.197
16:17:43.023 -> /BLOCK

16:17:43.023 -> BLOCK(299E):\
16:17:43.023 -> [0xFF][0xFF][0xFF][0xF][0xFF][0xFF][0xFF][0xF][0x0][0x0]\
16:17:43.023 -> [0x0][0x0][0x0][0x0][0x0][0x0][0x0][0x0][0x0][0x0]
16:17:43.023 -> [0x9][0x0][0x0][0xFF][0xFF][0xFF][0xF][0x8][0x9]\
16:17:43.023 -> [0x0][0x0][0x9][0x9][0x0][0x0][0xDF][0x10][0x0][0x0]\
16:17:43.023 -> [0xB][0x9][0x0][0x0][0xC][0x9][0x0][0x0]
[0x9]\
16:17:43.023 -> [0x0][0x0][0xE][0x9][0x0][0x0][0xF][0x9][0x0][0x0]\
16:17:43.023 -> [0x10][0x9][0x0][0x0][0x11][0x9][0x0][0x0][0x12][0x9]\
16:17:43.023 -> [0x0][0x0][0x13][0x9][0x0][0x0][0x14][0x9][0x0][0x0]\
16:17:43.023 -> [0x15][0x9][0x0][0x0][0x16][0x9][0x0][0x0][0x17][0x9]\
16:17:43.023 -> [0x0][0x0][0x18][0x9][0x0][0x0][0x19][0x9][0x0][0x0]\
16:17:43.057 -> [0x1A][0x9][0x0][0x0][0x1B][0x9][0x0][0x0][0x1C][0x9]\
16:17:43.057 -> [0x0][0x0][0x1D][0x9][0x0][0x0][0x1E][0x9][0x0][0x0]\
16:17:43.057 -> [0x1F][0x9][0x0][0x0] [0x9][0x0][0x0]![0x9]\
16:17:43.057 -> [0x0][0x0]"[0x9][0x0][0x0]#[0x9][0x0][0x0]$[0x9][0x0][0x0]%[0x9][0x0][0x0]&[0x9]\
16:17:43.057 -> [0x0][0x0]'[0x9][0x0][0x0]([0x9][0x0][0x0])[0x9][0x0][0x0]*[0x9][0x0][0x0]+[0x9]\
16:17:43.057 -> [0x0][0x0],[0x9][0x0][0x0]-[0x9][0x0][0x0].[0x9][0x0][0x0]/[0x9][0x0][0x0]0[0x9]\
16:17:43.057 -> [0x0][0x0]1[0x9][0x0][0x0]2[0x9][0x0][0x0]3[0x9][0x0][0x0]4[0x9][0x0][0x0]5[0x9]\
16:17:43.057 -> [0x0][0x0]6[0x9][0x0][0x0]7[0x9][0x0][0x0]8[0x9][0x0][0x0]9[0x9][0x0][0x0]:[0x9]\
16:17:43.057 -> [0x0][0x0];[0x9][0x0][0x0]<[0x9][0x0][0x0]=[0x9][0x0][0x0]>[0x9][0x0][0x0]?[0x9]\
16:17:43.057 -> [0x0][0x0]@[0x9][0x0][0x0]A[0x9][0x0][0x0]B[0x9][0x0][0x0]C[0x9][0x0][0x0]D[0x9]\
16:17:43.057 -> [0x0][0x0]E[0x9][0x0][0x0]F[0x9][0x0][0x0]G[0x9][0x0][0x0]H[0x9][0x0][0x0]I[0x9]\
16:17:43.090 -> [0x0][0x0]J[0x9][0x0][0x0]K[0x9][0x0][0x0]L[0x9][0x0][0x0]M[0x9][0x0][0x0]N[0x9]\
16:17:43.090 -> [0x0][0x0]O[0x9][0x0][0x0]P[0x9][0x0][0x0]Q[0x9][0x0][0x0]R[0x9][0x0][0x0]S[0x9]\
16:17:43.090 -> [0x0][0x0]T[0x9][0x0][0x0]U[0x9][0x0][0x0]V[0x9][0x0][0x0]W[0x9][0x0][0x0]X[0x9]\
16:17:43.090 -> [0x0][0x0]Y[0x9][0x0][0x0]Z[0x9][0x0][0x0][[0x9][0x0][0x0]\[0x9][0x0][0x0]][0x9]\
16:17:43.090 -> [0x0][0x0]^[0x9][0x0][0x0]_[0x9][0x0][0x0]`[0x9][0x0][0x0]a[0x9][0x0][0x0]b[0x9]\
16:17:43.090 -> [0x0][0x0]c[0x9][0x0][0x0]d[0x9][0x0][0x0]e[0x9][0x0][0x0]f[0x9][0x0][0x0]g[0x9]\
16:17:43.090 -> [0x0][0x0]h[0x9][0x0][0x0]i[0x9][0x0][0x0]j[0x9][0x0][0x0]k[0x9][0x0][0x0]l[0x9]\
16:17:43.090 -> [0x0][0x0]m[0x9][0x0][0x0]n[0x9][0x0][0x0]o[0x9][0x0][0x0]p[0x9][0x0][0x0]q[0x9]\
16:17:43.090 -> [0x0][0x0]r[0x9][0x0][0x0]s[0x9][0x0][0x0]t[0x9][0x0][0x0]u[0x9][0x0][0x0]v[0x9]\
16:17:43.090 -> [0x0][0x0]w[0x9][0x0][0x0]x[0x9][0x0][0x0]y[0x9][0x0][0x0]z[0x9][0x0][0x0]{[0x9]\
16:17:43.125 -> [0x0][0x0]|[0x9][0x0][0x0]}[0x9][0x0][0x0]~[0x9][0x0][0x0][0x7F][0x9][0x0][0x0][0x80][0x9]\
16:17:43.125 -> [0x0][0x0]/BLOCK

[... 230 lines cut ...]
16:22:04.467 -> G1 X73.086 Y112.795 E531.51895
16:22:04.467 -> G1 X72.457 Y112.103 E531.5345
16:22:04.467 -> G1 X71.641 Y111.86 E531.54866
16:22:04.467 -> G1 X64.58 Y111.861 E531.66609
16:22:04.467 -> G1 X63.761 Y112.104 E531.68029
16:22:04.467 -> G1 X63.406 Y112.459 E531.68864
16:22:04.467 -> G1 X63.125 Y112.801 E531.696
16:22:04.467 -> G1 X62.901 Y113.556 E531.7091
16:22:04.467 -> G1 X63.127 Y114.45 E531.72444
16:22:04.467 -> G1 X63.768 Y115.072 E531.73929
16:22:04.467 -> G1 X64.589 Y115.26 E531.7533
16:22:04.467 -> G0 F3600 X64.789 Y115.26
16:22:04.467 -> G0 X64.789 Y115.76
16:22:04.505 -> G0 X64.364 Y117.25
16:22:04.505 -> ;TYPE:SKIN
16:22:04.505 -> G1 F1800 X71.867 Y117.25 E531.87807
16:22:04.505 -> G1 X/VOL_BLOCK_CUR
```

`15.07 E531.47562` => 
`16:22:04.358 -> G1 X72.452 Y115.07 E531.4/VOL_BLOCK_PREVPREV`
+
`16:22:04.467 -> VOL_BLOCK_CUR :7562`

Again: data is normal, but parsing is wrong.

sapzxc, 2021-05-24:
I found the problem and created pull request with bugfix.
Desynchronization of command queue happens when fatGet failed inside seekSet after `if (card.errorCode())`. Result of execution setIndex not checked. As result in next call of get_sdcard_commands we read data from old position endOfLineFilePosition but file position SdBaseFile::curPosition_ did not changed. 
This error happens when position of new block of FAT placed in another cluster, so we need to call fatGet to read structure for seekSet and this operation fails and curPosition_ stay not changed.

This error also exists in original Ultimaker2Marlin repo.

gr5, 2021-05-24:
Wow.  This error is probably in the S3, S5, UM2+C printers as well!  @TinkerGnome - what are your thoughts?

gr5, 2021-05-24:
@sapzxc - have you rebuilt and are you printing with your bugfix now?

sapzxc, 2021-05-24:
Tomorrow will check with my glitch card. Now it printing model and I can not stop it :o)

gr5, 2021-05-24:
Glancing at the code change - I assume this only applies to SD card reading and not USB reading, right?  And it's only for error recovery if Marlin reads bad data from the SD card?  If so this won't affect UM3/S3/S5/UM2+C printers.  Only UMO/UM2 series printers.

sapzxc, 2021-05-24:
Logs explanation here.

Block 1303F break with "G1 X72.034 Y16.897 E1" then reading next block and has error. Next time we read "97" because "seek" position wrong.

```
16:30:53.261 -> BLOCK(1303F):G1 X60.602 Y16.897 E18.12997
16:30:53.261 -> G1 X60.794 Y16.897 E18.13316
16:30:53.261 -> G1 X60.602 Y17.089 E18.13496
16:30:53.261 -> G1 X60.464 Y17.258 E18.13774
16:30:53.261 -> G1 X60.13 Y17.665 E18.14649
16:30:53.301 -> G1 X59.744 Y18.969 E18.16911
16:30:53.301 -> G1 X60.135 Y20.511 E18.19556
16:30:53.301 -> G1 X61.261 Y21.604 E18.22166
16:30:53.301 -> G1 X62.542 Y21.897 E18.24351
16:30:53.301 -> G1 X69.864 Y21.897 E18.36528
16:30:53.301 -> G1 X71.137 Y21.599 E18.38702
16:30:53.301 -> G1 X72.256 Y20.511 E18.41298
16:30:53.301 -> G1 X72.649 Y18.966 E18.43949
16:30:53.301 -> G1 X72.245 Y17.626 E18.46276
16:30:53.301 -> G1 X71.886 Y17.231 E18.47164
16:30:53.301 -> G1 X71.583 Y16.897 E18.47914
16:30:53.301 -> G1 X72.033 Y16.897 E18.4832
16:30:53.301 -> G1 X72.034 Y16.897 E1/BLOCK

16:30:53.301 -> filepos=8170, sd_count=0, c=[A, ]
16:30:53.301 -> filepos=8171, sd_count=0, c=[47, G]
16:30:53.301 -> filepos=8172, sd_count=1, c=[31, 1]
16:30:53.301 -> filepos=8173, sd_count=2, c=[20,  ]
16:30:53.301 -> filepos=8174, sd_count=3, c=[58, X]
16:30:53.301 -> filepos=8175, sd_count=4, c=[37, 7]
16:30:53.301 -> filepos=8176, sd_count=5, c=[32, 2]
16:30:53.301 -> filepos=8177, sd_count=6, c=[2E, .]
16:30:53.301 -> filepos=8178, sd_count=7, c=[30, 0]
16:30:53.301 -> filepos=8179, sd_count=8, c=[33, 3]
16:30:53.301 -> filepos=8180, sd_count=9, c=[34, 4]
16:30:53.301 -> filepos=8181, sd_count=10, c=[20,  ]
16:30:53.301 -> filepos=8182, sd_count=11, c=[59, Y]
16:30:53.301 -> filepos=8183, sd_count=12, c=[31, 1]
16:30:53.301 -> filepos=8184, sd_count=13, c=[36, 6]
16:30:53.337 -> filepos=8185, sd_count=14, c=[2E, .]
16:30:53.337 -> filepos=8186, sd_count=15, c=[38, 8]
16:30:53.337 -> filepos=8187, sd_count=16, c=[39, 9]
16:30:53.337 -> filepos=8188, sd_count=17, c=[37, 7]
16:30:53.337 -> filepos=8189, sd_count=18, c=[20,  ]
16:30:53.337 -> filepos=8190, sd_count=19, c=[45, E]
16:30:53.337 -> filepos=8191, sd_count=20, c=[31, 1]

16:30:53.337 -> BLOCK(299E):\
16:30:53.337 -> [0xFF][0xFF][0xFF][0xF][0xFF][0xFF][0xFF][0xF][0x0][0x0]\
16:30:53.337 -> [0x0][0x0][0x0][0x0][0x0][0x0][0x0][0x0][0x0][0x0]
16:30:53.337 -> [0x9][0x0][0x0][0xFF][0xFF][0xFF][0xF][0x8][0x9]\
16:30:53.337 -> [0x0][0x0][0x9][0x9][0x0][0x0][0xDF][0x10][0x0][0x0]\
16:30:53.337 -> [0xB][0x9][0x0][0x0][0xC][0x9][0x0][0x0]
[0x9]\
16:30:53.337 -> [0x0][0x0][0xE][0x9][0x0][0x0][0xF][0x9][0x0][0x0]\
16:30:53.337 -> [0x10][0x9][0x0][0x0][0x11][0x9][0x0][0x0][0x12][0x9]\
16:30:53.337 -> [0x0][0x0][0x13][0x9][0x0][0x0][0x14][0x9][0x0][0x0]\
16:30:53.337 -> [0x15][0x9][0x0][0x0][0x16][0x9][0x0][0x0][0x17][0x9]\
16:30:53.371 -> [0x0][0x0][0x18][0x9][0x0][0x0][0x19][0x9][0x0][0x0]\
16:30:53.371 -> [0x1A][0x9][0x0][0x0][0x1B][0x9][0x0][0x0][0x1C][0x9]\
16:30:53.371 -> [0x0][0x0][0x1D][0x9][0x0][0x0][0x1E][0x9][0x0][0x0]\
16:30:53.371 -> [0x1F][0x9][0x0][0x0] [0x9][0x0][0x0]![0x9]\
16:30:53.371 -> [0x0][0x0]"[0x9][0x0][0x0]#[0x9][0x0][0x0]$[0x9][0x0][0x0]%[0x9][0x0][0x0]&[0x9]\
16:30:53.371 -> [0x0][0x0]'[0x9][0x0][0x0]([0x9][0x0][0x0])[0x9][0x0][0x0]*[0x9][0x0][0x0]+[0x9]\
16:30:53.371 -> [0x0][0x0],[0x9][0x0][0x0]-[0x9][0x0][0x0].[0x9][0x0][0x0]/[0x9][0x0][0x0]0[0x9]\
16:30:53.371 -> [0x0][0x0]1[0x9][0x0][0x0]2[0x9][0x0][0x0]3[0x9][0x0][0x0]4[0x9][0x0][0x0]5[0x9]\
16:30:53.371 -> [0x0][0x0]6[0x9][0x0][0x0]7[0x9][0x0][0x0]8[0x9][0x0][0x0]9[0x9][0x0][0x0]:[0x9]\
16:30:53.371 -> [0x0][0x0];[0x9][0x0][0x0]<[0x9][0x0][0x0]=[0x9][0x0][0x0]>[0x9][0x0][0x0]?[0x9]\
16:30:53.371 -> [0x0][0x0]@[0x9][0x0][0x0]A[0x9][0x0][0x0]B[0x9][0x0][0x0]C[0x9][0x0][0x0]D[0x9]\
16:30:53.371 -> [0x0][0x0]E[0x9][0x0][0x0]F[0x9][0x0][0x0]G[0x9][0x0][0x0]H[0x9][0x0][0x0]I[0x9]\
[... 18 lines cut ...]
16:30:53.445 -> G0 X62.677 Y20.697
16:30:53.445 -> G1 F1800 X69.726 Y20.697 E40.21004
16:30:53.445 -> G1 X70.54 Y20.507 E40.22394
16:30:53.445 -> G1 X71.176 Y19.887 E40.23871
16:30:53.445 -> G1 X71.404 Y18.993 E40.25405
16:30:53.445 -> G1 X71.174 Y18.232 E40.26727
16:30:53.485 -> G1 X70.545 Y17.54 E40.28282
16:30:53.485 -> G1 X69.729 Y17.297 E40.29698
16:30:53.485 -> G1 X62.668 Y17.298 E40.41441
16:30:53.485 -> G1 X61.847 Y17.541 E40.42865
16:30:53.485 -> G1 X61.493 Y17.896 E40.43698
16:30:53.485 -> G1 X61.213 Y18.238 E40.44434
16:30:53.485 -> G1 X60.989 Y18.993 E40.45743
16:30:53.485 -> G1 X61.215 Y19.887 E40.47277
16:30:53.485 -> G1 X61.855 Y20.509 E40.48761
16:30:53.485 -> G1 X62.677 Y20.697 E40.50163
16:30:53.485 -> G0 F3600 X62.877 Y20.697     <-------- 97 here
16:30:53.485 -> G0 X62.877 Y21.197
16:30:53.485 -> /BLOCK

16:30:53.485 -> filepos=8170, sd_count=0, c=[39, 9]
16:30:53.485 -> filepos=8171, sd_count=1, c=[37, 7]
16:30:53.485 -> filepos=8172, sd_count=2, c=[A, ]
16:30:53.485 -> insert buf(bufindw=2, bufindr=5, buflen=5): 97
```

sapzxc, 2021-05-24:
> Glancing at the code change - I assume this only applies to SD card reading and not USB reading, right? And it's only for error recovery if Marlin reads bad data from the SD card? If so this won't affect UM3/S3/S5/UM2+C printers. Only UMO/UM2 series printers.

I never used another 3D printers or another versions of Ultimaker so I do not know how it works on another models.

sapzxc, 2021-05-25:
Tested my pull request #113 with my bad SD Card. Now it works great. Printing works as designed. Serial console show reading errors but repeated reading works great and printing process does not stop.

Looks like fix inside SdBaseFile.cpp also exists in all projects that using official Arduino SdFat Library.

Models will never screw up when your SD card has small problems with reading. Only when a lot of continuous errors occurred you will see fatal error on screen "SD card error" and printing process will be stopped.

closed 2021-05-26 by sapzxc