TinkerGnome/Ultimaker2Marlin #114: Pause mid print and large retraction

frez-knee, 2021-09-08:
UM2+ with Bondtech feeder, running Tinker V19.03.1. 

During a print from SD card, the print will pause and do a large retract, around 800mm or so. Then it will feed the same length in. The feeder will either lose grip and fail to reload the material, resulting in a failed print or due to the length of the feed, it will deposit a blob on the print and continue printing as normal. 

Troubleshooting/fixes I've tried: 
- this behavior does not happen during USB prints
- cleaned the SD card slot with compressed air and cleaned contacts with IPA
- swapped Ulticontroller board with my other UM2+ Mark2 (running Tinker Dual V19.03.1) that does not have this issue

None of the above helped. After reverting to factory FW modified by Bondtech, I have not seen it happen. Which leads me to believe the FW was the issue.

gr5, 2021-09-08:
The problem is a bad bit in the ascii gcode.  So there will be something like
G1 X123.455 Y93.212 E777.111
7 is an ascii 55 but more importantly the last 3 bits are "111" which is 7 in binary.  If any of those 3 bits get switched to a zero it will retract (100mm, 10mm, 1mm respectively) and then on the next gcode it will correctly go back to where it was.

Other common errors is that the X or Y axis will move 10 or 100mm out of the print and then go back into the print continuing where it left off.

But the root of the problem can be many things.  I had this issue on one printer and it got MUCH better when I cleaned out the SD card slot but it can be caused by the ribbon cables under the printer or many other things.  I have 19.03.1 on 3 active UM2 printers but this issue only happens to me on one of them and much rarer now that I've cleaned it out (once very 100 prints or so).

frez-knee, 2021-09-08:
Hey gr5, thanks for the response. You probably don't recognize my screen name, but we've discussed this before in another github issue (#101). 

Cleaning did seem to help, but still much more frequent than once every 100 prints like you achieved. If the problem is related to the SD card slot, I thought it was strange the problem did not follow with the Ulticontroller board after swapping it. 

I only have two active UM2+'s, but like you mentioned this retraction issue only occurs on one of them even though they were both running the same V19.03.1 FW.

gr5, 2021-09-08:
It's more likely to be in the circuit board that holds the SD card.  Or maybe a cross talk problem with the 2 ribbon cables (how they are routed - try to separate them by another 5mm if possible).

I tried swapping that circuit board with another UM2 and the problem went away but didn't appear on the other printer - at least not for the next year.  Now it's starting to happen on a different printer but still much to rare to try to fix it.

gr5, 2021-09-08:
If you remove that "sd" card board (probably called the "display" board) then there is a trick to get the knob off - you can push it off from behind with a screw driver or other skinny tool.

frez-knee, 2021-09-08:
> It's more likely to be in the circuit board that holds the SD card. Or maybe a cross talk problem with the 2 ribbon cables (how they are routed - try to separate them by another 5mm if possible).
>
> I [...]

That's what I'm referring to when I say the SD card slot, the [SD card connector/reader](https://www.mouser.com/ProductDetail/Hirose-Connector/DM1AA-SF-PEJ72?qs=q%252B4FLHukgWDRiUKW8ueLgA%3D%3D). It's connected to the Ulticontroller/Display Board, which is what [fbrc8](https://fbrc8.com/collections/ultimaker-2-and-ultimaker-2-extended/products/ulticontroller-board) calls it. 

I did consider cross talk between the cables a possibility, but since they're routed the same way on both of my UM2+'s, I figured it was more related to an issue with SD reader itself. 

> If you remove that "sd" card board (probably called the "display" board) then there is a trick to get the knob off - you can push it off from behind with a screw driver or other skinny tool.

Yes, I know. As mentioned, I removed both boards and swapped them. There's a small slotted hole in the back of the clear plastic piece to push the knob out with an allen key, etc.