TinkerGnome/Ultimaker2Marlin #120: Machine occasionally unloads filament at the end of a print run 18.11.1

JeffDM2, 2023-05-30:
I'm running 18.11.1 on my Ultimaker 2 Go. Once in a while when a print finishes, the machine just unloads the filament. I run the same program multiple times and it doesn't always do it. I thought the "G1 E-5" did it as if maybe the zero length didn't work, and commented it out but it still seems to happen as often.

Here is my ending g code:

G28 X0 ; home the X-axis
M104 S50 ; turn off heaters
M140 S75 ; turn off bed
G92 E0 ; zero the extruded length
G92 E0 ; zero the extruded length
;G1 E-5 F1000 ;retract
M84 ; disable motors

gr5, 2023-05-30:
Did you adjust the flow rate?

I'm pretty sure there is a bug that if you increase flow rate live while printing from the TUNE menu (or on the geek mode printing screen) then when it gets to the end of the print it gets a bit confused about the current position of filament and it retracts quite a bit.  Basically if you print 5 meters of filament at 110% flow rate that would be 10% X 5 or .5 meters of position confusion.

If you set flow to < 100% then I assume it would EXTRUDE a lot at the end of the print.

gr5, 2023-05-30:
It's pretty easy to accidentally touch the dial while printing and change the feedrate by 1% without noticing (I've done it by accident).  But I suspect you did not change the feedrate.

In which case I'm not sure what causes this bug but I've seen it (very rarely) - I have 18.03 on one um2go and 19.03 on another um2go - both are pretty active.  I haven't noticed the problem in several years.

There is a different, simliar problem where the printer does random retracts throughout the printer.  Maybe every 2 mintues, maybe every 100 minutes.  It would typically show up as the printer seemingly pausing for 20 seconds randomly and then continuing for no reason.  Or failing the last few layers with the filament retracted out of the feeder.  This final symptom - could that be what you see?  It's a hardware issue and quite fixable if you have that issue.

JeffDM2, 2023-05-30:
I will try to pay more attention for future runs and see if that’s what’s going on.

sapzxc, 2024-01-15:
See my fix at https://github.com/TinkerGnome/Ultimaker2Marlin/issues/112. It's a problem related to SD card FAT processing. After implementing my fix, I haven't encountered this bug at the end of printing for 2 years. However, I did experience your problem before applying the fix.

gr5, 2024-01-16:
Unfortunately, although your pull request was merged, tinker hasn't built a version of the firmware since your PR.