TinkerGnome/Ultimaker2Marlin #122: Ulticontroller dead, need replacement

acca84, 2023-11-02:
Hello,
I'd like to know if I can use btt 12864 mini with this firmware?
Thanks

gr5, 2023-11-03:
I really don't know.  If it works with regular "ultimaker marlin" firmware then it will work with this version as well.  I googled around a bit but didn't learn a lot.  It might work.  Did you break your display?

acca84, 2023-11-03:
I've just bought an um2 with ulticontroller dead, I'm thinking about solutions to save it

gr5, 2023-11-03:
I'd probably get a duet.

So the ulticontroller is on 2 different boards - the display PCB and the main PCB.  The main PCB has the "arduino" and the servo controllers and does almost everything and then the display PCB just has the knob and the display and the USB flash drive slot.

How did you isolate where the problem is?

Could it be the power brick?

Do the lights come on?

Did you connect a USB "printer" cable to the back to talk to the main PCB to make sure it can move the servos and such?  Maybe we should talk about this over on the ultimaker forum.

gr5, 2023-11-03:
Just to clarify: the 12864 is just a display, right?

acca84, 2023-11-03:
The motherboard works with cura connected to USB
The 12864 is a module including screen, sdcard reader and knob
I think it is supported by marlin or at least mks marlin branch, I'll check that

gr5, 2023-11-04:
There's a small chance you will have to swap 2 pins.  I vaguely remember that someone changed a display and 2 of the pins were backwards.  Seems crazy.  But I don't remember the details and it was many years ago.