TinkerGnome/Ultimaker2Marlin #124: G4 (Dwell) should not send action:pause / action:resume

Aimeethyst, 2024-01-18:
G4 is used for brief pauses in movement during a print (e.g. Simplify3D makes a slight pause before bridging to ensure a solid connection). Sending action:pause suggests to controlling software that the printer is requesting a pause, so for example OctoPrint will run any configured pause and resume scripts everytime a bridge is started.

gr5, 2024-01-18:
Are you posting to the wrong open source project?  This fork of Marlin only runs on ultimaker 2 series printers.  I don't understand what "action:pause" is.  Is this a gcode command?  Is this something internal to klipper, marlin, or octoprint?

Aimeethyst, 2024-01-18:
Thanks for responding, sorry, I should have made it clearer in the report.

Commit 02f7869dd14a430e346354ccc2b884fbb7c1e6a6 way back in 2016 added `serial_action_P(PSTR("pause"));` and a corresponding resume when handling G4 in `void process_commands()`

gr5, 2024-01-30:
@TinkerGnome - please click the link below to see the code in question.  In the commit Aimee links to just above, you added messages that go back through the USB cable to the host that send messages to the host which could be for example Prontrface, Cura, octoprint or something else.

You added these "pause" "resume" messages in several places and then later commented them all out except the one for G4. **Do you remember why you did this?  Do you print over serial cable?**  I don't so this doesn't affect me.

So for Aimee who probably uses S3D and octoprint, S3D is inserting these G4 commands for bridging and this is sending a message to Octoprint which then runs a "pause script" which is not desired when trying to do bridging as it probably moves the Z up a few mm as well.

@Aimeethyst - you could do a pull request with these two serial_action_P commands commented out to make Tinker's job a little easier.  He'll likely pull it but motivating him to create a new release is harder.

https://github.com/TinkerGnome/Ultimaker2Marlin/blob/0ae8deb7190cdd7d37a6329f633dff226d21d1f5/Marlin/Marlin_main.cpp#L1198C1-L1214C13