TinkerGnome/Ultimaker2Marlin #126: 3P-bed leveling Z-adjust

Berend1953, 2025-02-09:
I am rebuilding a Ultimaker 2+ in a Laserengaving/cutting machine.
This means a different toolhead with a much lower tool zero pont.

Tool zeropoint can be adjusted with M206  set additional homeing offset.
Problem is the Bedleveling, the 3 point bedleveling has apparently fixed points and does not move with the M206 offset.

How to change the Z-value of the points in the Three point Bed leveling procedure?
I did al lot of searching but could not find it.

Thanks in advance. 

Berend

gr5, 2025-02-09:
I don't think you can - I suspect any leveling code never made it into Ultimaker's version of Marlin but I'm not certain.  But the UM2 test bed is so stable, you only have to manually level once every few months.  less than that if you con't change nozzles (well, "tools" in your case).

Berend1953, 2025-02-09:
Thanks for your super fast response.
It is what it is.